WildRikku/OctoPrint-SpoolManager #2: When printing immediately, sometimes filament amount is not calculated

WildRikku, 2024-09-09:
When clicking "load & print" from a file, sometimes, the filament amount is not calculated and the according error is displayed. I suspect a race condition because (so far) it never happened when loading first and it is always fixed by loading a different file, loading the file again and then starting the print.

WildRikku, 2024-09-10:
I think I figured it out. Octoprint does some processing of uploaded files. When the files are large, this can take several seconds. You can tell by the dropdown arrow being faded out directly after upload and becoming usable after a moment. Once it's usable, the problem never occurs when clicking "load and print". So, a workaround is to wait. SpoolManager might be able to wait until the file is processed though before it throws an error. One could also argue that Octoprint should not load the file if it's not processed yet.

WildRikku, 2024-09-11:
While Octoprint updates the estimated time when the file has been processed, SpoolManager does not update the filament amount needed, so it should react to the end of the processing. You can see that when uploading a huge file and loading it. The status section of Octoprint will show almost nothing at first, but when the file is processed, it will be updated.

WildRikku, 2024-10-17:
Installing PrintTimeGenius "solves" the issue for some users because it speeds up the calculation of the needed filament significantly.

WildRikku, 2024-10-17:
Since processing an uploaded file can take a significant amount of time, just waiting with no notice is not an option. I opted for making the warning more precise so users can tell that they have to wait for the file.

closed 2024-10-21 by WildRikku

barrelltitor, 2025-04-10:
In case others search here, using octoprint with orcaslicer will always result in this error unless you're on the latest version or apply [this commit](https://github.com/OctoPrint/OctoPrint/commit/f4fbe9193856e828e5c83444be0cab7acedcb50d) because the actual analysis fails no matter what you do

It might be a good idea @WildRikku to state that there might be an error with the analysis and to check the logs, as I tried all potential fixes from here before I realized "oh wait of course there's no metadata, the analysis fails".

I also made a [temporary repo](https://github.com/barrelltitor/octoprint-temp-fix) with the fixed files so people can just mount them if needed