WildRikku/OctoPrint-SpoolManager #3: Old issues that might still be relevant

WildRikku, 2024-09-11:
https://github.com/OllisGit/OctoPrint-SpoolManager/issues/154

https://github.com/OllisGit/OctoPrint-SpoolManager/issues/50

https://github.com/OllisGit/OctoPrint-SpoolManager/issues/39

https://github.com/OllisGit/OctoPrint-SpoolManager/issues/18

https://github.com/OllisGit/OctoPrint-SpoolManager/issues/5

https://github.com/dojohnso/OctoPrint-SpoolManager/issues/38 (this could be done by having a checkbox "calculate offset automatically", which both activates the offset and deactivates the offset text fields - just need to figure out when to update this, as the offset is added to Gcode commands and adding an offset is actually different from having the same temperature all the time. The poster probably comes from a point where the Gcode sets temp only once - in the beginning.)

Bug: https://github.com/dojohnso/OctoPrint-SpoolManager/issues/47

https://github.com/dojohnso/OctoPrint-SpoolManager/issues/51

https://github.com/dojohnso/OctoPrint-SpoolManager/issues/53

https://github.com/dojohnso/OctoPrint-SpoolManager/issues/55

numericOverflow, 2024-09-13:
Hey, @WildRikku - glad to see someone picking up this add-on!

I would still love to see the absolute temp override tackled, so I migrated it from old repo as ticket #9 over here.

I'm often lurking around GH, so feel free to ask if you have questions about the issue!

Hottabych-, 2024-09-14:
>https://github.com/OllisGit/OctoPrint-SpoolManager/issues/5
This was already solved by [Webguyatwork]https://github.com/Webguyatwork and merged into [dojohnso ](https://github.com/dojohnso) work. I'll use it successful since February in my printer farm. It's broken now.

As always I offer to provide a German translation, but I need a file with the original English text, because I have no clue to work with git.

WildRikku, 2024-09-14:
> > [OllisGit#5](https://github.com/OllisGit/OctoPrint-SpoolManager/issues/5)
> > This was already solved by [Webguyatwork]https://github.com/Webguyatwork and merged into [dojohnso ](https://github.com/dojohnso) work. I'll use it successful since February in my printer farm. It's broken now.
> 

I saw the pull request and will check it out!

> As always I offer to provide a German translation, but I need a file with the original English text, because I have no clue to work with git.

Thank you! The plugin currently has no translation workflow yet, so I can't easily provide you with a list of strings. I'd have to go through all the files manually and exchange strings with function calls. It's on my mind for the future but it will take a while to get there.

Did you use Python or JavaScript before, just not with git, or are you not a coder at all? If you have a little experience with coding, you could probably help me implement it.

Hottabych-, 2024-09-14:
I coded a lot professionally in Assembler, Pascal, Fortran 77, C, and some other, but ... this is ~ 30-40 years ago ;-) Object oriented languages where in their infancy at that time. In the last years until today I've done some rare hobby projects in C but I'm very far away from my old experience. So I'm not really a help for you. I can read & understand languages of today but that's it. If you are ready for translation or if you have another idea - I can help in that topic.

>I saw the pull request and will check it out!
Thank's a lot because my sqldb access is broken with the update, I had one central db for my printers only and I've a big problem now.

WildRikku, 2024-09-14:
> I coded a lot professionally in Assembler, Pascal, Fortran 77, C, and some other, but ... this is ~ 30-40 years ago ;-) Object oriented languages where in their infancy at that time. In the last [...]

Alright, that's fine. :) Feel free to open a separate issue for translation here to make it easier for you to check if I am working on it. This ticket here will be closed at some point. Otherwise, I'll hopefully remember to ping you when I start working on it. :)

Hottabych-, 2024-12-19:
>marked for closing
sure?