WildRikku/OctoPrint-SpoolManager #9: Use absolute temperatures instead of offset temperature overrides

numericOverflow, 2024-09-13:
Migrated this issue/request from [old maintainer repo](https://github.com/dojohnso/OctoPrint-SpoolManager/issues/38):

Is is possible to have SpoolManager submit the absolute temperatures stored on spool profile instead of the offset temp? 

I often want to print something in a new material, but I have to first manually calculate the difference between the original temp set in the GCODE and update the spool's offset.  It would be **MUCH** easier if I could just set a desired temp for the spool and have it submit that temp as an absolute override temp rather than an offset.

If that's not possible for some reason (say if you can only send offsets and not absolute values), could you dynamically calculate the offset on the fly and send the calculated offset?  Just subtract the GCODE's temp(s) from the spool's absolute temp and send that value?  Maybe even have a toggle button in the spool config screen to choose which value is used for that spool (absolute or offset temps).

[image]

WildRikku, 2024-09-13:
Hey, thanks for migrating. I'll note my initial thoughts down here for now. I like the idea and will probably implement it in some way, I'll just have to see how difficult it is and when I have time.

Do you ever print files that change temperatures mid-print? I don't but I can imagine those cases might be a little tricky to handle.

Initially I thought this could be done by having a checkbox "calculate offset automatically", which both activates the offset and deactivates the offset text fields - just need to figure out when to update this, as the offset is added to Gcode commands and adding an offset is actually different from having the same temperature all the time.

Bootscreen, 2024-09-24:
This Feature would be so awesome.

I print some models like the Benchy with every new Filament. And it makes it so much easier when i can set the needed temps per spool directly in octoprint without the need of resliceing.