WildRikku/OctoPrint-SpoolManager #10: External SQL DB Connection deleted!

Hottabych-, 2024-09-14:
Thank you for take over the responsibility. 
You don't fork 1.7.4.dev0; now I lost my SQL db data connection with all my printer. Please fork the correct https://github.com/dojohnso/OctoPrint-SpoolManager/releases/tag/1.7.4.dev release! 
--> This is a huge issue for all my installations.

WildRikku, 2024-09-17:
Hi, I did not fork the dev branch because it had conflicting commits with the main branch, so I couldn't get it into the 1.7.4 release quickly. Other users also reported that the functionality was broken anyway. I guess you used the dev channel. I can setup an unstable branch so you can get the dev channel back; I currently cannot guarantee that it will work though. Migrating the external database code is on my to do list for the stable channel but to have it there I need to test it myself.

Hottabych-, 2024-09-18:
Hi,
>I guess you used the dev channel.

yes you are right.

>I can setup an unstable branch so you can get the dev channel back;

That would be a good idea. I will check that out when available. But for bigger tests I'v time earliest in 1,5 weeks cause I've a hard ans busy time at work during this period. So please don't expect an early answer if you are quicker. At the moment I've stopped all print activities not to mess my data up.

>Other users also reported that the functionality was broken anyway.

I don't understand their issues, because I haven't had any with that branch. This might have to be digged into to understand the differences.

Thanks!

Hottabych-, 2024-11-30:
Hi, how is the status of this topic?

WildRikku, 2024-12-20:
I have just pushed a pre-release that contains the external database support. It is mostly untested and unchanged from the original code in the old repository. I was able to setup a MySQL database and connect to it, please see the release notes for hints. I don't usually use MySQL or Postgres, that's why it took a while until I got around to at least briefly test the code after merging it.

I will close this ticket as the code is now out there. Please open a new one for any bugs.

closed 2024-12-20 by WildRikku