WildRikku/OctoPrint-SpoolManager #13: Database Sharing

ramosglauco, 2024-09-18:
Hi. I have 2 printers connected to the same raspberry. Each printer runs its own instance of octoprint. I would like to share the same SpoolManager database for both printers. Is this possible? Thx!

WildRikku, 2024-09-19:
Hi, this is currently not possible via the UI. It is also untested if it works at all.

If you need this quickly and are familiar with the Linux shell, you could try:

1. setting up SpoolManager on both instances of OctoPrin (no spools, just installation),
2. shutting down both instances,
3. then on the Pi shell deleting the database file of SpoolManager of one instance and replacing it with a symlink to the other database: Note how the file is called (it's in SpoolManager settings under storage), delete it and then `ln -s <path to other database> <previous database name without path>`.

Please note that the database file you delete will be permanently lost.

Also please note that this is untested. It _should_ work, but it will cause issues if both instances of OctoPrint try to write to the database of SpoolManager at the same time, because the plugin will not know that it's sharing the file. If you try this, please let me know if it worked. Please make backups.

ramosglauco, 2024-09-19:
Unfortunately I don't have the necessary knowledge to do this myself. Would it be possible for you to provide the option to share the same database link directly in the plugin UI?

Sorry for the simplistic question, but what would happen if this link were editable and I put exactly the same one for my other instances? For exemple

Spool Manager storage menu in my Octoprint1 
[image]

Spool Manager storage menu in my Octoprint2
[image]

WildRikku, 2024-09-19:
I have just recently adopted the plugin, so I don't know how easy it would be to make that field editable and actually work. I think it's possible though - it would result in the same behaviour as the suggestion I made for the command line.

I have marked this ticket as enhancement suggestion and will look into it. I think it would be useful. :)

ramosglauco, 2024-09-19:
I suggest placing a checkbox to allow editing of the field. And leave a warning that this is something "beta" for now.

Thank you for adopting this plugin. It is extremely useful, especially for those who work with large-scale 3D printing like me.

WildRikku, 2024-09-20:
It's not done with making the text field editable. It's currently without function. I can make a beta version available as soon as I have actually programmed the functionality.

I will soon release a new stable version that will make a "testing" channel available. If you subscribe to that "testing" channel, you will get the feature in beta state as soon as I release it.

ramosglauco, 2024-10-01:
Hi. Any updates on this feature or a new version of the plugin?

WildRikku, 2024-10-01:
Not yet. Any updates will be posted on this issue or in release notes of published versions, which you will receive via OctoPrint.

ramosglauco, 2024-10-01:
ah, ok. Thx

WildRikku, 2025-01-07:
Update: Just editing the value in the field in the settings dialogue is not sufficient. The path is currently hardcoded in some places. The current testing release contains the option to use an external database (MySQL or PostgreSQL). The same external database can be used by different instances of OctoPrint. This feature has experienced very little testing, so if you test it, please read the release notes and make backups. I'd appreciate any feedback though.

I still plan to make it possible to use a local database with multiple OctoPrint instances in the future, but it will not be as simple as I hoped.

turboc1208, 2025-07-09:
I'm getting an error that the mysql drivers are not installed.  How did you get around that?

WildRikku, 2025-07-13:
> > I'm getting an error that the mysql drivers are not installed. How did you get around that?
> 
> This depents on the Operating System where your Octoprint Instance is installed on. Ask an AI Chatbot from Google or Microsoft, they will tell you how to install the missing mysql driver for your operating system.

AI chatbots are no tools for searching information. Also, this does not have anything to do with mysql drivers for an operating system. The operating system only determines how Python packages are installed.

> I'm getting an error that the mysql drivers are not installed. How did you get around that?

You need a Python package supplying the MySQL libraries. Since you use Docker, you will need to add these packages to your Docker containers. If you need help, please ask there where you got the Docker files from.

turboc1208, 2025-07-13:
I got it working today.  The OS is important in that there are different drivers needed for windows than for linux.  But that's beside the point.  I ended up opening up a shell under the docker container so that I was in it's filespace.  Then I used apt to  install python3-pymysql.  I'm not sure I needed that one.  But then I used pip to install pymysql.

apt install python3-pymysql
pip install pymysql

Then I started getting an error about a specific user not being able to login.  I changed to loggin in as root and the test worked.  you will have to setup the mysql database link and save it.  Then restart that octoprint instance.  That will create the objects it needs in the database.  Then you can use the button to copy the local copy to the mysql database.  But be warned.  It truncates the table and overwrites whats in there.  So pick which one you want in there and copy it.  Don't copy the second one.  I copied one I had with about a dozen spools in it, and then copied my other instance which only had one spool in it.  When I looked I only had one spool.  At least it doesn't destroy the local database so I could go back and recopy that one.  

We'll see how it handles both systems printing.  One thing I think I'm going to want is, the ability to see where a spool is being used if it's currently printing something and I want to use it on another printer.