WildRikku/OctoPrint-SpoolManager #14: Re-select spool on filament runout

gozergozarian, 2024-09-20:
Hi, this could just be something I haven't figured out how to accomplish, but when a spool is expended and I change it mid-print, it would be great to be able to re-select a spool. 

I don't know what, if anything the printer reports back to Octoprint when the printer pauses due to filament runout so this could be complicated or impossible. 

I am running a Prusa MK3S+ on 3.14 firmware.

WildRikku, 2024-09-21:
Do you print from SD card or do you start your print from OctoPrint? How does the printer know that the filament is empty, does it have a sensor or do you use an OctoPrint plugin?

gozergozarian, 2024-09-22:
I print from OctoPrint and the printer has a sensor.

WildRikku, 2024-09-22:
> I print from OctoPrint and the printer has a sensor.

Did OctoPrint notice when the print paused (did it offer to continue)? OctoPrint has functionality to react to paused prints but these are not supported by all printers.

gozergozarian, 2024-10-22:
Sorry for the delay! I didn't get a chance to reproduce this until now. OctoPrint stays in state: printing but the Print Time does not continue to tick up. Here's the "State" area while the printer has paused for filament reload: 
[image]

gozergozarian, 2024-10-22:
It is a Prusa, and yeah I think they have less support for OctoPrint in favor of their self-developed option.

You do understand correctly - after the filament is changed, OctoPrint shows progress and the print continues without any additional interaction with OctoPrint. 

I only recently started using this plugin, so I have a bunch of partial spools with unknown filament left on them. Any spool that I don't know the contents of I print using a spool I created called "Junkspool" that I don't track usage on. 

I think I would be totally happy with a situation where I manually pause OctoPrint, have a way to interact with SpoolManager to select a different spool, and then resume OctoPrint.

WildRikku, 2024-10-22:
Alright. Asking for a spool change when pausing from OctoPrint was also something I thought of. That would require you to pause OctoPrint when your printer auto-pauses, but it would also work for people with other printers. I will check if that is technically possible.