WildRikku/OctoPrint-SpoolManager #16: [BUG] [1.7.4] No Spool selected on every print

Bootscreen, 2024-09-24:
Hello,

since v1.7.4 i get the message that i didn't selected a spool when i start to print, even though i selected a spool. The spool selection itself seems to work, because it counts the used amoung of filament from the spool.

greetings
Bootscreen

WildRikku, 2024-09-24:
Hi,
could you please provide the exact text of the message you get? Also, do you print from SD card or from OctoPrint?

If you print from OctoPrint, please try starting a print that you have already completed in the past and let me know if the error occurs for that print, too. There is a known issue with printing files that have recently been uploaded.

When you say it counts the filament, do you just tell the plugin to print anyway even though it thinks that no spool was selected? (it's fine if it works, I just need more details.)

Bootscreen, 2024-09-26:
ok, looks like it only happens for the first print. Now on new prints (old and new files) it recognize the selected spool. i think it can be closed.

but for the completeness: I print from octoprint.

and yes. i selected the spool, ignores the warning that i doesn't select a spool and it counts down the selected spool.

Bootscreen, 2024-09-26:
i close it as can't reproduice and didn't happen again