WildRikku/OctoPrint-SpoolManager #17: [Feature Request] Don't spam "SPM:WARNING: Filament not enough!" warnings

NewsGuyTor, 2024-09-28:
Every time I open Octoprint to check print process on a print, started on minus measured filament (but still plenty to complete the print), I get a warning and notification.

I don't see the point of that, considering I consciously disregarded all the warnings of that exact thing, at the start of the print.

Please note that I don't want to hide the warning popup and notification before starting a print, I want to hide the notification that appears every single time a new Octoprint tab is opened during printing.

https://github.com/OllisGit/OctoPrint-SpoolManager/issues/354

WildRikku, 2024-09-28:
For completeness, please name your OctoPrint version, SpoolManager version and which browser you use.

I can confirm the issue and consider the re-appearing notification a bug. I'll need to figure out if this is related to OctoPrint or the plugin directly. For now, an obvious workaround would be to adjust the remaining filament amount on your spool, as it's clearly not correctly calculated if there is sufficient filament even though SpoolManager does not say so :) Though you probably have already considered that option.

NewsGuyTor, 2024-12-28:
.

commit c3b4de9f (2025-01-02): Introduce a new way to suppress warnings so the user does not get spammed. Fixes #17. (also add a little unrelated TODO comment)
closed 2025-01-02 by WildRikku