WildRikku/OctoPrint-SpoolManager #18: Attention: Needed Filament could not be calculated

duncanlovett, 2024-10-13:
Hello,

I have been using the plug in fine until today. Whenever I try to print a file I get the message:

"Attention: Needed Filament could not be calculated (missing Metadata or spool -fields)
There is no spool selected for tool ) despite it being used (Maybe) by this print.
Do you want to start the print without a selected spool?"

I have tried deleting and re-installing the plugin but despite it  showing a spool is selected, I get this message. I have since tried printing another gcoded file and everything works ok. What could be the issue with my Prusa Sliced Gcode file that would cause this.

I enclose the gcoded file renamed to by a txt file for uploading here.

Many thanks

Duncan

[LED_Profile_V3_3h47m_0.28mm_205C_PLA_ENDER3V2.gcode.txt](https://github.com/user-attachments/files/17356844/LED_Profile_V3_3h47m_0.28mm_205C_PLA_ENDER3V2.gcode.txt)

WildRikku, 2024-10-14:
Hello, I will check if I can find the issue. Please specify your SpoolManager plugin version, OctoPrint version and how you installed OctoPrint (for example, by using OctoPi). A screenshot would also be useful (so I can see SpoolManager in the OctoPrint sidebar).

duncanlovett, 2024-10-16:
Will do many thanks

duncanlovett, 2024-10-17:
Hello,

It's:
OctoPrint Version 1.10.2
Installed by Octopi Image and then updated.
SpoolManager Plugin (1.7.4)

Screen  shots attached.

Thank you

[image]
[image]
[image]
[image]

WildRikku, 2024-10-17:
I have re-opened the issue as you probably did not mean that the problem was solved.

Thank you for providing information. I can not see something obvious so I have more questions.

a) Do you use a printer with more than one extruder?

b) Could you please check if the filament that causes the error has density and diameter set? If not, that is the reason and you can solve the issue by setting those fields. If they are already set, I suspect that something is weird with your database. In that case I would like to find out what and why.

For that, I would like to ask you for a copy of the database. This will make debugging easier. Please note that it may contain information like the spools you use of course including price, where you bought them etc, so if you are not comfortable sharing that, I can understand.

The database is stored on your Pi in the home dir under `.octoprint/data/SpoolManager/spoolmanager.db`. You can upload it here (that will be public), upload it to a file sharing service and post the link here or send it to me: `spoolmanager@wildrikku.de`. If you do not know how to copy the file, you will find help on the Octoprint Discord server.

duncanlovett, 2024-10-17:
Sorry must have hit the wrong button.

It's an Ender 3 v2 with a Sprite Pro Extruder so single extruder.
Will double check the spool settings and mail you the file if it's not that. 

Many thanks

duncanlovett, 2024-10-17:
No that looks ok, will send you over the database.
Thanks

[image]

duncanlovett, 2024-10-17:
I've now tried another file that has worked previously with a roll that also has previously worked, but its not doing the same thing :-(

[image]

duncanlovett, 2024-10-17:
It might be related or not, but I ignored the warning and carried on printing and now that print is not listed in Print Job History either.

WildRikku, 2024-10-17:
Hmm. I received your file. I imported it and tried it with the gcode you provided above and could not reproduce the error, neither with the transparent filament nor with any other filament. I have two more things that might help.

a) Since you made an export of your SpoolManager data that works for me, you could try re-importing your own export, using the "replace" option. I would recommend doing that when no print is running.

b) To rule out any coincidence with another plugin, please check if the error persists when OctoPrint is started in safe mode. You can find instructions on how to do so here: https://docs.octoprint.org/en/master/features/safemode.html

Regarding Print Job History, that plugin was previously maintained by the same person as SpoolManager and has some integration with it, so it makes sense it fails to collect data when SpoolManager fails to collect data. We just need to find out why SpoolManager fails to collect data.

duncanlovett, 2024-10-17:
I will try safe mode first and then re-importing and see if that fixes it. I have tried deleting the plugins and restart, re-install but with no luck.

Thanks again.

WildRikku, 2024-10-17:
> I will try safe mode first and then re-importing and see if that fixes it. I have tried deleting the plugins and restart, re-install but with no luck.
> 
> Thanks again.

Sorry, I made a mistake in my suggestion. OctoPrint safe mode does not allow enabling plug-ins at all, so you won't be able to test in safe mode. I thought you could start in safe mode and then enable SpoolManager but you can't do that. So to check if another plug-in is causing the issue, you would have to disable them all manually.

duncanlovett, 2024-10-17:
I will try that out tomorrow as I have a large overnight print running currently.
Thanks again.

duncanlovett, 2024-10-18:
Ok, so I used the same file but with a different filename and uploaded it to Octoprint. I hit print and got the same error. I waited a few seconds then hit cancel. I then waited a few seconds and hit print and up came the box saying do you want to start the print with the correct filament. I hit cancel and every time I hit print it works. I tried the original file again and it came up with the error, so I waited then ht cancel. I waited and tried again but still the original error. This happens every time I repeat it. I have tried renaming the file incase the file name was too long or had some character in it, but the same error. I have deleted it and now re-uploaded it and it seems fine, so wonder if somehow it was corrupt? 

Thanks again for your help.

WildRikku, 2024-10-18:
> Ok, so I used the same file but with a different filename and uploaded it to Octoprint. I hit print and got the same error. I waited a few seconds then hit cancel. I then waited a few seconds and [...]

Alright thanks for trying again. This sounds very much like the issue I had. So just to make sure I understand it correctly: The file that worked after waiting is the exakt same file that does not work?

Also, the file that still does not work, even with waiting - is that the one you printed anyway?

I would assume that if you avoid printing and use the waiting workaround, the issue should not come up with newly uploaded files.

duncanlovett, 2024-10-18:
Yes to both of your statements. I have now uploaded a totally new gcode for another item and that works fine, so it must have been something to do with the file name I guess?

Still at least things are working again, I'll try to use more basic shorter filenames and let you know if I have any further issues.

Thanks again for you help and a great plugin!!

WildRikku, 2024-10-18:
Alright. I am not so sure about the filename, I also had some cases where I suspected that the file analysis, which is what you wait for with the workaround, breaks when you force the print. But I could not yet confirm that.

Good to hear you are currently not having issues. If you encounter any problems again, please open a new ticket and reference this one. Happy to help and thanks for your kind words!

--

Closing this as duplicate of #2.

closed 2024-10-18 by WildRikku

WildRikku, 2024-10-18:
@duncanlovett 
Regarding manual editing, can't mail you for some reason:

Yes, absolutely. Just edit your spool and manually set the "used" fields. SpoolManager does nothing special when it counted the used filament.

If you have any more questions feel free to join the OctoPrint Discord. I and several other people are often around to help.

duncanlovett, 2024-10-21:
Just to say, everything is working fine now and I've done another 20 or so prints with zero issues.

Thanks again for your help @WildRikku