WildRikku/OctoPrint-SpoolManager #19: Multi-color spools

andrewroth, 2024-10-17:
Hi, has there been any discussion on adding support for multi-color filaments? For two-colors, I would envision a line diagonally in the color square, with half one color and half the other. Could do similar for tri-colour. UI would have to be updated to support selecting a second and possibly third color. I would consider implementing it, though I don't want to commit 100% yet.

WildRikku, 2024-10-18:
There has been no such discussion to my knowledge, no, and there are currently no plans for an implementation. I personally don't use such spools. Do they have a defined length per color? Would you simply like to have a multi-color square or also actual functionality?

andrewroth, 2024-10-18:
I had in mind solely multi-color square, no other functionality. Though it would be handy to have a true boolean or field for which kind of filament. I would categorize it as dual-color, tri-colour, or rainbow. Rainbow does have color per length, but it's not important to know the exact distance, just that it's rainbow. In fact, just a preset "rainbow icon" would be sufficient for me to know visually it's a rainbow. Maybe a dark and light version, as they seem to come in dark and light rainbows versions.

For dual/tri-colour filament, it's merged right into the filament and is uniformly merged for the whole spool. Ex:

[image]

In those cases, depending what angle you look at the filament, you see a different color. Having the box divided into two or three parts and you choose the color you want in each is what I had in mind.

I will also add, these filaments come in silk and regular matte PLA. There's already a PLA_silk Material type, and it could visually just throw a vertical light reflection down the icon. Ex. 
[image]. I think that could be done with a png with transparency so it can apply to any filament color box. It would be very nice to visually see which filaments are silk or not when I go to choose which one to print. There's even some ex. green that can be a solid matte green, or green silk, and you have to rely on reading the name currently. Not such a bad thing, but a visual indication would be even better.

WildRikku, 2024-10-19:
Alright, thanks for the details. I agree that having a split color box would be neat with up to three or even four colours and a "rainbow" option. It could be done by having a "add color" button which splits the color field into the next amount (so, in two when clicked first, in three the next time). It's not that easy though because it requires changes to the database so those colors can be stored, too (the database stores hexadecimal color values). Also the dropdown needs to be able to handle those multi-colors.

The rainbow could be a specially treated value in the dropdown which turns the color field into a CSS gradient or an icon. 

If you want to make a pull request, I'd be happy to help. I am currently busy with the external database support and some bugs (and the part of my life that is not volunteering), but if you provided a UI implementation, I could take care of how to store the data in the database.

Sembazuru, 2024-11-05:
Not just rainbow or co-extrusion. Check out Amolen, they have lots of different color change filaments with different ranges of colors. Having multiple colors like an "add color" button as suggested by WildRikku would be useful.

Jcodeerd, 2024-12-22:
Something where you could define multiple colours and split up the box would be nice, I don´t think there are filaments where there is a defined length of each colour. Most filaments I bought are X colours mixed together

ianrifkin, 2025-03-19:
I have seen (and own) different types. I have a "rainbow" spool that literally is one color for a length then another color for a length etc. and then I also have blended types where there are two colors more blended throughout the length of the spool.

In an ideal implementation in Spool Manager it would support listing every color in the spool BUT I would happy if in the dropdown there was just a "multicolor" option that changed the square from a solid color to a rainbow or something like that and then we just list the colors in the text field.

Currently I list the multiple colors in the text field and that works but when glancing the spools it would be nice if there was a visual indication that it's a multi-colored spool as currently it requires choosing one solid color to be displayed in the square.