WildRikku/OctoPrint-SpoolManager #28: 1.8. alpha / error in Pluginmanager

Hottabych-, 2024-12-23:
When installing the 1.8. alpha I got a strange behavior in the plugin manager:
"There was an error retrieving the plugin data from the backend. Please report this in the bug tracker with all requested logs."

[image]

No installed plugins to see any more. As you can see the list is empty. This seems to me being critical. I don't want to use the bug tracker because I think this error was generated by your plugin, because just before installation it was all fine.

2nd: the copy feature from internal db to external db din't work when new installed the 1.8 alpha. 
The internal db have suddenly the old entries at the time you switch to your new branch. Hopefully the user have an backup/export of the internal db (like me). Then he can import this csv file once more to the internal db and then the copy feature work and replace the old entries in the external db.  BTW I use MySQL.

This were the rough results of a 'just install and play' session.
In the next 2-3 weeks I have to check whether printing and spool management will work ... but not before Christmas ;-) and not direct after theses days...my family wants to have my full attention.

WildRikku, 2025-01-01:
The error message is pretty clear. There was a problem getting plugin data from the plugin backend. This is not related to SpoolManager because SpoolManager is not related to the backend. Either there was a temporary issue or your installation is corrupted. If the problem persists, try asking on the Octoprint Discord. You should also actually collect the logs requested in the OctoPrint bug tracker and provide them.

WildRikku, 2025-01-01:
For the second problem, please open a second issue (please do not put multiple issues in one) and provide a step-by-step process to reproduce the problem, logs and more details on your setup (packages, version numbers, operating system). I tested the copying and it worked for me. I can't help you when I don't know your entire setup and setup process.

closed 2025-01-01 by WildRikku

Hottabych-, 2025-01-05:
Update to the new version (1.8.0a1) no problem, work like a charm. With the update the first error is gone now. Maybe there was a hickup in the installation. 
At the moment I only use one of my printer withe the new plugin for test reasons.  If there is something I can test send me an message.