WildRikku/OctoPrint-SpoolManager #29: Where database is located?

den1251, 2024-12-28:
I broke my system, so I don't have access to octoprint UI directly. So, if it is possible how can I extract the database from the filesystem to use it later? What file extension/name database has? I think if it is possible it is useful to add it to the bottom of documentation. (If somebody else will be as lucky as me 😁)

WildRikku, 2024-12-30:
It depends on how you installed Octoprint, but in general, there should be a .octoprint folder in your homedir and data/SpoolManager/ in it and the spoolmanager.db file in there is your database.

den1251, 2024-12-30:
Thank you! (I just forgot to show hidden files and that's why I wasn't able to find it the first time)

closed 2024-12-30 by den1251