WildRikku/OctoPrint-SpoolManager #33: Updating spoolsManager to version 1.8 or version 1.7 start showing a new user trying to be save into my google key vault

rabbit3dcustom, 2025-01-22:
Hi,

Every time i restart the octoprint instances with versions "Version 1.7.7 (bug fix release)" or "Version 1.8.0a1 (testing release)", there is a new user "Olli" with password "il####" trying to be saved in my google key vault, the autofill password stuff.

It seems to be something from "Olligist" the former owner, but this raise all my alerts and warnings about being hacking, not good :( 

Reinstalling version "1.7.6" seems to fix it.

Could you please, take a look at it?

Thank you so much.

Ajimaru, 2025-01-22:
See here [https://github.com/search?q=repo%3AWildRikku%2FOctoPrint-SpoolManager+%22Olli%22&type=code](https://github.com/search?q=repo%3AWildRikku%2FOctoPrint-SpoolManager+%22Olli%22&type=code)
I think, this is not concerning.

rabbit3dcustom, 2025-01-22:
Nice! glad to see that the passwords are the same and in plain text... but, as i wrote, something is raising to my chrome the user and password and it advice me to save it to the keyvault.

That is not happening in previous versions.

WildRikku, 2025-01-28:
As you have correctly guessed, this is code that was already in the plugin before I took it over, so it can't be caused by an update. These are just example strings. However, you did not mention which page uses those credentials, so I cannot help you. Also please specify what you mean by "Google key vault". Is that where your browser stores passwords? Is it a browser native function or a browser extension?

Ajimaru, 2025-03-12:
> As you have correctly guessed, this is code that was already in the plugin before I took it over, so it can't be caused by an update. These are just example strings. However, you did not mention [...]

The example is used in the configuration for a external database as username and password. I guess Chrome grabs theese vaules and try to save it for the users convenience.

[image]

WildRikku, 2025-03-12:
If Chrome tries to save those credentials, that's because it's the default and you did not configure it otherwise. The default is that Chrome asks you if it should save the password. You can tell Chrome to not save credentials for certain sites. This has nothing to do with hacking and it is also not caused by the plugin.

If you don't use a MySQL database anyway, you don't need the 1.8.0a1 testing version and should use 1.7.7 stable instead.

I have however removed the default credentials from the code so the text fields will be empty in the future.

closed 2025-03-12 by WildRikku