WildRikku/OctoPrint-SpoolManager #34: SpoolManager Update 1.7 to 1.8 repeating itself

AlexVercammen, 2025-01-26:
Morning,
OctoPrint prompts to update SpoolManager 1.7 to 1.8. Once updated, it prompts to update 1.8 to 1.7 and so forth.
To break the cycle, I just had to ignore the 1.7 update. I am just letting you know.
I am running OctoPrint v1.10.3, OctoPi v0.18.0, running on Raspberry Pi 3 Model B Rev 1.2

lsellens, 2025-01-28:
I have the same issue. It prompts to update to the 1.8.0a1 build, but once installed to prompts again to install the 1.7.7 stable build and the cycle keeps repeating. I don't think the version is properly tags as a non stable build if I had to guess. I do not have testing releases selected in octoprint but for what ever reason it doesn't realize 1.8.0a1 is a testing release until after its installed.
https://github.com/cp2004/OctoPrint-Knowledge/blob/main/release-channels.md
https://docs.octoprint.org/en/master/bundledplugins/softwareupdate.html#version-checks

lsellens, 2025-01-28:
[image]

WildRikku, 2025-01-28:
Did any of you ever install the 1.7.7 update? If not, please do so, then upgrade to 1.8.0a1 and check if 1.7.7 is still suggested. Make sure to restart OctoPrint after both updates.

WildRikku, 2025-01-28:
I can't reproduce this issue with my own production installation. Please load updates without cache (button in the toolbar on top), install 1.8.0a1 and restart OctoPrint and see if 1.7.7 is still suggested.

@AlexVercammen Have you ever installed 1.7.7?

sdh2, 2025-01-30:
I am having the same issue with 1.7.7 and 1.8.0a1. I also have the release channel selected.

AlexVercammen, 2025-01-30:
I had 1.7.7 installed when OctoPrint notified me of the update. Once update to 1.8.0 the cycle started. So, 1.7.7 was re-installed then 1.8.0. When OctoPrint restarted again, it notified me of the update 1.7.7 and I selected "ignore" instead of update. And all is fine as I can tell.
Merci.

mholzma, 2025-01-30:
Just wanted to add that I am having the same issue.

numericOverflow, 2025-01-30:
Same issue here.

LMK if there's some debug info you want or would like me to try.

```
Octoprint:  Version 1.11.0rc1
SpoolManager:  1.7.7  & 1.8.0a
Running on a Raspi 4, just built it from the latest octopi image about 2 weeks ago, so everything is very current.
```
I constantly get the notification 1.7.7 -> 1.8.0a1 ->1.7.7... loop  update notice from octoprint notifications.  Both claim to be updates, and the same thing happens regardless of having 1.7.7 or 1.8.0.a1 installed, then just alternate back and forth.

Then as soon as octoprint reboots, I get prompted with an update (even when it's a downgrade) the "other" version as the update:

The plugin manager showing currently installed version is on top, software updates page is below.  When I click the "check updates" the dialog appears prompting me to update to the "other" version.    I've gone through the loop a bunch of times, but nothing seems to stop it.

[image]

I captured the output from the update from 1.7.7 to 1.8.0a1 if that helps?  It's pretty boring so I won't include it here.

WildRikku, 2025-01-31:
Hmm. I am currently rather busy but will try to reproduce the issue when I get to work on the plugin the next time. This is curious, as what you all describe is an issue I had myself and was pretty sure I fixed. I'll probably ask for more information here; for now, you should know that it is unproblematic to ignore the update once you have the version installed that you want. I know it's annoying, but it will not cause any further trouble.

WildRikku, 2025-01-31:
Actually, I noticed something in the screenshot that made me very sure I had simply not included the fix in the release and that seems indeed to be the case. I don't know why my own installation did not fall into the update loop, but however, I have just pushed out a 1.8.0a2 release that should end the update loop.

For those interested, the issue was that the 1.8.0a1 release would not see itself (and other testing releases) and hence suggest the newest version it knew, 1.7.7, as an update.

You will probably not see the 1.8.0a2 update as long as you are on 1.8.0a1 so please install 1.7.7 one more time, it should then suggest 1.8.0a2 which should then end the update loop. You might have to switch to the testing channel again after installing 1.7.7.

numericOverflow, 2025-01-31:
Well, still sort of broken, but closer.... though I think I figured out why:

Release Channel = "Only Release" ==> Broken
- Loops between 1.7.7 & 1.8.0a2 yet

Release Channel = "Devel RC" ==> Works! 
- 1.7.7 updates to 1.8.0a2 and stops thinking 1.7.7 is an update to cause loop.

[image]

Ajimaru, 2025-01-31:
> Well, still sort of broken, but closer.... though I think I figured out why:
> 
> Release Channel = "Only Release" ==> Broken
> 
>     * Loops between 1.7.7 & 1.8.0a2 yet
> 
> 
> Release Channel = "Devel RC" ==> Works!
> 
>     * 1.7.7 updates to 1.8.0a2 and stops thinking 1.7.7 is an update to cause loop.
> 
> 
> [image]

I can confirm this observation, same behaviour on my installation.

WildRikku, 2025-01-31:
There is no "Devel RCs" but from your screenshot I can tell what you mean ("Devel RCs" is Octoprint's channel).

I changed something, please do "check for updates (no cache)" and let me know if 1.7.7 stops suggesting 1.8.0a2 when set to "Only Release".

WildRikku, 2025-01-31:
The following process worked for me: From 1.8.0a2, set to "Only Release". OctoPrint will suggest to install 1.7.7. After doing so, no further updates are suggested.

rtdog, 2025-02-01:
I confirm that WildRikku's process to upgrade to 1.8.0a2 followed by 1.7.7 worked for me.  As a side note, numericOverflow's method also worked, but I undid that and went with WildRikku's method.  Thank you for this great software!

WildRikku, 2025-02-01:
> I confirm that WildRikku's process to upgrade to 1.8.0a2 followed by 1.7.7 worked for me. As a side note, numericOverflow's method also worked, but I undid that and went with WildRikku's method. Thank you for this great software!

Thank you for your feedback! :)

numericOverflow, 2025-02-02:
> The following process worked for me: From 1.8.0a2, set to "Only Release". OctoPrint will suggest to install 1.7.7. After doing so, no further updates are suggested.

Confirmed - this works for me too.  As of now, 1.7.7 release/stable stopped suggesting update to the 1.8.x release candidate.

Thanks for the fix.

WildRikku, 2025-02-02:
Alright, thanks for the feedback. I will consider this ticket closed then.

closed 2025-02-02 by WildRikku

Jake-Mac-210, 2025-03-08:
> The following process worked for me: From 1.8.0a2, set to "Only Release". OctoPrint will suggest to install 1.7.7. After doing so, no further updates are suggested.

This worked for me too..

FINALLY THE ANNOYING "UPDATE AVAILABLE" B.S. MESSAGE IS GONE!!!!!!