WildRikku/OctoPrint-SpoolManager #36: Update hell: 1.7.7 to 1.8.0a2

rtrout, 2025-02-08:
When I'm OctoPrint update manager it prompts me to update to 1.7.7, and when on 1.7.7 it prompts me to update to 1.8.0a2

Is this expected? Should update manager be prompting to move to a test release? Should the test release prompt to reverse to standard release? Or is something borked with my config, and I shouldn't be getting prompts for any test releases?

WildRikku, 2025-02-08:
No, this is not intended. Considering this issue has been reported before and then reported as fixed by several users, I am unsure what's happening. The bug that caused this behaviour is not present in 1.7.7 or 1.8.0a2.

Did you try checking updates with cache ignoring? And which channel do you have selected? Please share screenshots of your plugin update view in OctoPrint for both versions.

WildRikku, 2025-02-08:
possible duplicate or regression of #34

Maliron, 2025-02-13:
On 1.8.0a2 and it's still asking me to "upgrade" to 1.7.7 still saddly.

rtrout, 2025-02-16:
Sorry @WildRikku for taking so long to reply. Thank you to others for reporting seeing the same symptoms.

> Did you try checking updates with cache ignoring? And which channel do you have selected? Please share screenshots of your plugin update view in OctoPrint for both versions.

I have Release Channel set to 'Only Release' while on 1.8.0a2, and I'm prompted to update to 1.7.7.

For screenshot, is this what you wanted?:

[image]

If so, I can 'update' to 1.7.7 and repeat.

I'm not sure how to 'check updates with cache ignoring'???

rtrout, 2025-02-16:
Updating with screenshots.

When on 1.8.0a2 (with Update Now button showing):

[image]

When on 1.7.7, the OctoPrint notification:

[image]

And on 1.7.7 the Software Update screenshot:

[image]

Apart from answering about 'check updates with cache ignoring' I think I've answered your questions.

robotpandarocket, 2025-02-24:
same for me

fortepc, 2025-02-25:
Having the same issue, screenshots proving the loop

[image]

[image]

[image]

[image]

[image]

WildRikku, 2025-03-01:
Since nobody explicitly mentioned trying that, please try loading updates without cache. It's an option in the dropdown button in the plugin update manager. I can't reproduce the problem and I also can't find any errors in the code or the setup.

It might also help to uninstall and re-install SpoolManager. In that case, remember to make a backup of your database. It depends on how you installed Octoprint, but in general, there should be a .octoprint folder in your homedir and data/SpoolManager/ in it and the spoolmanager.db file in there is your database.

rtrout, 2025-03-02:
I'm still unclear on the instructions for how to 'try loading updates without cache'. I've tried googling with no luck. Am I looking in the right place? Screenshot below with green question marks to suggest where I might need to look.

[image]

WildRikku, 2025-03-02:
@rtrout On your screenshot, there's a dropdown arrow right of "Check for updates". It's hardly visible because of your dark theme.

rtrout, 2025-03-02:
If I follow you correctly, I select 'Force check for updates':

[image]

and get this:

[image]

I'm a bit reluctant to do the backup/uninstall/reinstall, may have time later in the week.

WildRikku, 2025-03-02:
Oh wow, it does not say "ignores Cache" in English. Sorry for that.

Could you please do the following to give me more insight:
1. Ignore the update, change the channel to testing and "force check for updates" again and let me know if it still suggests 1.7.7 (it should not)
2. Go back to "Only Release", "force check for updates" again, expecting it will suggest 1.7.7 and install that, then with 1.7.7 installed and "Only Release" channel active, "force check for updates" once more and let me know if it suggests any update. If not, it was a caching issue and the problem is solved. If it still wants to update to 1.8.0a2 on the "Only Release" channel, the cache is not the problem, which is what I wanted to find out.

rtrout, 2025-03-02:
I don't have a release channel option for 'testing', I have Only 'Release' or 'Release & Testing'.

[image]

Maliron, 2025-03-03:
I can confirm that on stable after rolling back to 1.7.7 it is no longer suggesting update. So it appears it was indeed a cache issue. Thanks for looking into this for us! 
Side note, is there a way to have it default to a tool when selecting a spool? Like many, I only have 1 tool, and when changing spools I have to remember to select the tool as well to get it to actually change spools.

WildRikku, 2025-03-03:
> I can confirm that on stable after rolling back to 1.7.7 it is no longer suggesting update. So it appears it was indeed a cache issue. Thanks for looking into this for us! Side note, is there a way [...]

Good to hear the issue is resolved for you. Regarding the tool selection, please open a separate issue, in which you describe what your steps are exactly (as there are multiple ways to select  a spool). Usually you don't have to select a tool when there is only one, so maybe it's a bug - let's find out on a separate issue.

aschmitz, 2025-03-22:
Just to be explicit about the apparent solution to this, I had the same experience (started on "Only Release", as far as I know I was only _ever_ on "Only Release", but got prompts to update to 1.8.0a2 and then back to 1.7.7 and repeat), but the following appears to have fixed it:

1. Get back to where 1.7.7 is installed if you're on 1.8.0a2.
2. In OctoPrint Settings, under Software Update, change the SpoolManager Plugin Release Channel to "Release & Testing".
3. Click the arrow next to "Check for updates" and choose "Force check for updates", ignore the prompted update to 1.8.0a2.
4. Change the Release Channel back to "Only Release".
5. Trigger "Force check for updates" again.
6. You should get a green message telling you everything is up to date, and not be prompted to install 1.8.0a2.

In particular, the "Force check for updates" does not seem to be sufficient once OctoPrint already thinks 1.8.0a2 is an available update, you seem to need to trigger a channel change to clear that information. (It is also possible that skipping step 3 is sufficient, I didn't try that.)

WildRikku, 2025-03-22:
So the Software Update Plugin that comes with OctoPrint keeps a cache and that only gets reset on channel switch. Good to know. That was probably why my instructions were insufficient for some people. Thanks @aschmitz

seraphx2, 2025-05-28:
But just to double check, will you be addressing how 1.8.0a2 got in the main release track in the first place?

WildRikku, 2025-06-01:
> But just to double check, will you be addressing how 1.8.0a2 got in the main release track in the first place?

The issue had been fixed minutes after releasing 1.7.7, but due to caching issues, several users were still affected.