WildRikku/OctoPrint-SpoolManager #38: Odd behavior with multiple filaments selected (MMU - Single tool)

JesNoFear, 2025-02-28:
I am seeing some odd behaviors with multiple filaments selected.

1. When print starts, it pulls spool 0 (T1) filament offset (yay) - but pulls the bed offset from the last spool selected
2. when tool changes occur, offsets are not updated. 
3. When print starts and T2 is the first tool, it still pulls filament offset for spool 0 (T1) and bed offset from the last spool listed (in my test case spool 2 - T3)

WildRikku, 2025-03-01:
Please post your SpoolManager and OctoPrint version.

JesNoFear, 2025-03-01:
Spoolmanager 1.7.7
Octoprint 1.10.3

WildRikku, 2025-03-02:
Alright, I'll look into it.

JesNoFear, 2025-03-03:
No problem,
here is the extruder configuration
[image]

here is what the slicer looks like,
[image]

here is a 3mf and gcode file
[V350-SS-HQ-MMU-Cute Bee Lg (0,1,3).zip](https://github.com/user-attachments/files/19061515/V350-SS-HQ-MMU-Cute.Bee.Lg.0.1.3.zip)

Tools refer to the toolhead, some printers may have multiple, but in a shared nozzle configuration, it refers to the filament channel on the MMU, they are called by gcode with T0, T1, T2, ... think of it as an automated M600.

this is probably the best layout i can show you,
[image]

T0 is what everyone uses, but when its the only one defined, there is no need to call it.

you are correct, when i said filament offset, i meant tool offset for the filament
[image]

if you start diving into the 3mf, it calls upon macros, i have attached them incase.
[Macros and Klipper Config.zip](https://github.com/user-attachments/files/19061655/Macros.and.Klipper.Config.zip)

in the toolchange gcode in the slicer, i call M109 expecting octoprint to adjust based on the offset, but the offset isnt changing after initially set.

[image]

Here is a snip-it of a toolchange in the gcode i sent you.
```; custom gcode: end_filament_gcode
; Filament-specific end gcode 
;END gcode for filament
; custom gcode end: end_filament_gcode
; custom gcode: toolchange_gcode
RESPOND PREFIX="info" MSG="Tool Change, Layer Height = 3.05"
G1 E-2 F1200
G0 Z5.05 F20000  ; move nozzle to remove stringing
G0 X215 Y357  ; parking toolhead xy
G0 Z4.5 ;  parking toolhead z

M109 S200  ; Tool Change Tool Temp

RESPOND PREFIX="info" MSG="Tool Change, Next Tool = 3"
T3 ; MMU Toolchange Command

RESPOND PREFIX="info" MSG="Tool Change, Next Tool = 3  Next Temp = 240"
M109 S240  ; set temp for now current tool for Octoprint to modify
CLEAN_NOZZLE
G0 X175 Y357  ; Safe zone
G0 Z5.05  ; current layer height
; custom gcode end: toolchange_gcode
; custom gcode: start_filament_gcode
; Filament gcode

; Macro syntax: https://help.prusa3d.com/article/macros_1775
; The "=~" operator uses regex enlosed by forward slashes (=~/regex/).
; The "==" operator can be used for an exact match instead if you wish (=="string").

; The below logic says:
; - IF printer settings profile name contains "Voron" (using regex)
; - AND nozzle size equals <value>
; - THEN insert the gcode (SET_ PRESSURE ADVANCE ADVANCE=<value>).
; Uncomment all lines below (and modify appropriately) if you wish to use this.

;SET_PRESSURE_ADVANCE ADVANCE=0.0495
;

; custom gcode end: start_filament_gcode
G1 E-1.000000 F1800
; stop printing object BeeLg.stl id:0 copy 7
EXCLUDE_OBJECT_END NAME=BeeLg_stl_id_0_copy_7
; printing object BeeLg.stl id:0 copy 0
EXCLUDE_OBJECT_START NAME=BeeLg_stl_id_0_copy_0
; acceleration to travel```

Let me know what other information you would like details about
Thank you for looking into this

JesNoFear, 2025-03-04:
> By "filament offset", you mean the "tool offset" configured in the filament?

a little more detail on this,

lets say i have 4 spools defined,

[image]

when i start a print, the tool offset will pull from selected spool 0

[image]

where the bed offset will pull from the last defined selected spool, in this case spool 3

[image]

the other two spools defined are

[image]

[image]

as i am stepping through this to capture screen shots, i actively see the bed offset being applied to bed and chamber on not the last spool defined in the list, but the last spool chosen, whether its spool 0, 1, or 8. if they are entered out of order its the last one chosen.

i wish i knew/understood python better so i could help. because now i am wondering are the calculations correctly subtracting from the proper spools as the print goes. I have been using partial / junk spools for the setup and calibration of the MMU so the counts are already off.

if it will help and/or if you feel up to it, we can setup a screen share on teams or discord if you would like.

WildRikku, 2025-03-04:
Thanks for all the additional info. I have an idea what might go wrong in the plugin code - there were issues with the bed temp before, which is why I asked for your plugin version. I'll check that in the next couple days and write again.

Regarding the amount of used filament, that's happening in a completely different part of the plugin and OctoPrint and I don't know of any issues there. If all your used spools show less filament after a print, I'd say it's probably reduced by the correct amounts. You could however check that quite easily by for example printing cubes using 1, 2 and 3 grams of filament (as calculated in the slicer) of one color each in the same gcode. Filament usage calculation is complex but should not be completely off.

WildRikku, 2025-03-06:
related https://github.com/OctoPrint/OctoPrint/issues/3144

WildRikku, 2025-03-06:
I can confirm all the issues both from observation and from looking into the code.

Now the thing is, after looking into the code, I figured out that all of the offset handling is handled by OctoPrint, not SpoolManager. SpoolManager does process GCode while printing but that's only for tracking the amount of used filament.

1. While SpoolManager can tell OctoPrint multiple offsets (one per tool), OctoPrint does not process this. I think this is a bug. I could confirm that multiple offsets are set correctly by SpoolManager for printers with real tools (multiple nozzles).  So the problem is that virtual tools do not work the same.
2. OctoPrint also seems to be responsible for starting with the offset of the first tool. It has to start with something but should change to the offset for T3 as soon as the T3 comes up in the GCode.
3. OctoPrint does not support multiple offsets for the bed. There is nothing SpoolManager can do about that (without a lot of effort).
4. The fact that the bed offset is set to the one configured in the last selected spool is caused by SpoolManager, but it's also caused by the fact that SpoolManager can only send one bed offset to OctoPrint. This means that if it could send multiple bed offsets, it would not matter which one it sets last.

I linked an OctoPrint feature request above, you might want to watch that. It's for offsets for virtual tools without SpoolManager.

I would suggest you open a bug report with OctoPrint for the tool offsets not changing mid-print for virtual tools (I would consider the first tool selection mid-print, too). Ping me there and I will provide additional info on how SpoolManager communicates with OctoPrint. Since I am not an OctoPrint developer and don't have a printer with multiple extruders and hence know very little about those setups, I think it makes more sense if you open the bug report.

You might also want to open a feature request for support of more than one bed temperature offset. I don't know what you need it for, so you might want to explain that there.