WildRikku/OctoPrint-SpoolManager #39: Python 3.12.9 Crash

CoppoloElectronics, 2025-03-02:
Installing octoprint_SpoolManager through the plugin manager causes Python 3.12.9 to crash after restarting the OctoPrint instance.

I am running Octoprint on macOS rather than Raspberry Pi, so this may be considered a non-standard install. That being said, Octoprint has been running 100% as expected prior to installing Spool Manager.

After installing through the plugin manager, I am prompted to restart Octoprint. Upon restarting, I am presented with a system dialog prompt to update the plugin. I have attempted to both bypass the update and perform the update and the result is the same. Upon subsequent Octoprint restart, the instance crashes and I get a system prompt that Python has crashed. I was able to locate a system problem report where it states that thread 28 has crashed:

`Thread 28 Crashed:
0   ???                           	000000000000000000 0 + 0
1   libz.1.3.1.zlib-ng.dylib      	0x000000011748fb80 inflateInit2 + 80
2   _imaging.cpython-312-darwin.so	0x00000001172409d5 ImagingZipDecode + 2133
3   _imaging.cpython-312-darwin.so	0x00000001171ead32 _decode + 82
4   org.python.python             	0x000000010e800304 method_vectorcall_VARARGS.llvm.13119612936639000081 + 124
5   org.python.python             	0x000000010e90bddc _PyEval_EvalFrameDefault + 46816
6   org.python.python             	0x000000010e7f686b method_vectorcall.llvm.6169543659237222586 + 383
7   org.python.python             	0x000000010e7f3eec _PyObject_Call + 275
8   org.python.python             	0x000000010e90dbe8 _PyEval_EvalFrameDefault + 54508
9   org.python.python             	0x000000010e7f67c7 method_vectorcall.llvm.6169543659237222586 + 219
10  org.python.python             	0x000000010e90dbe8 _PyEval_EvalFrameDefault + 54508
11  org.python.python             	0x000000010e7f68da method_vectorcall.llvm.6169543659237222586 + 494
12  org.python.python             	0x000000010ea128c4 thread_run + 126
13  org.python.python             	0x000000010e999434 pythread_wrapper.llvm.7693561602469046384 + 43
14  libsystem_pthread.dylib       	0x00007fff60059661 _pthread_body + 340
15  libsystem_pthread.dylib       	0x00007fff6005950d _pthread_start + 377
16  libsystem_pthread.dylib       	0x00007fff60058bf9 thread_start + 13`

I'd like to be able to use this plugin because it looks very helpful. For now, I've had to remove the following two folders from my site-packages to get Octoprint functional again:
octoprint_SpoolManager
OctoPrint_SpoolManager-1.7.7.dist-info

If any further information is needed, I'd be happy to provide it. Thanks!

WildRikku, 2025-03-03:
I don't have a Mac, but I will check if the problem might be caused by the specific Python version you have installed on your Mac. While I investigate, could you please elaborate on the following:

- What do you mean by "system dialog prompt to update the plugin"? An OctoPrint notification or a Mac notification? When exactly does that appear?
- Do I understand correctly that OctoPrint starts once after you installed the plugin, but after another restart, it no longer does? It might be helpful if you once again wrote what exactly happened in which order, just to rule out any misunderstanding.

CoppoloElectronics, 2025-03-03:
Thanks for looking into it.

> What do you mean by "system dialog prompt to update the plugin"? An OctoPrint notification or a Mac notification? When exactly does that appear?
This is after the initial installation of the plugin through the plugin manager. Octoprint shows the prompt that an update is available.

> Do I understand correctly that OctoPrint starts once after you installed the plugin, but after another restart, it no longer does? It might be helpful if you once again wrote what exactly happened in which order, just to rule out any misunderstanding.
Correct.
- As per usual plugin installation, I am prompted to restart Octoprint.
- When I restart, I am prompted (by Octoprint) that an update is available for the plugin.
- **IF** I update the plugin, I am prompted to restart Octoprint again.
- Upon restart, Octoprint does go through the startup sequence and as soon as it reaches the dashboard, I am met with a macOS system alert that Python has crashed.
- Even if I do not go through with the Octoprint prompted update as mentioned two steps ago, I am met with the same fate if I restart the Octoprint instance.

I hope this helps explain things more clearly!

CoppoloElectronics, 2025-03-03:
Thanks for the information. I may take a look into alternative Python options. The current stable Octoprint release (1.10.3) is not compatible with Python 3.13, so I opted to use the last stable version of Python 3.12.x when I set up the server.

If/when I adjust my Python version and have a chance to test, I'll report back with any updates.