WildRikku/OctoPrint-SpoolManager #40: Must select tool every time when changing spools.

Maliron, 2025-03-03:
When selecting a new spool from the Spool Manager tab on the main window, I must hit the dropdown for tool and choose Tool 0 in order to be able to hit "Select for Printing."

[image]

If I use the the little dock box next to the main window by clicking the ... then it seems to default to Tool 0.

[image]

And, I've only just now realized that the tool is actually a saved parameter for each spool, for people who have multiple tools and may have colors or types assign to them, and if I set the tool and hit save then it will default to that every time for that spool, and now I feel like a tool. XD

I will still post this just in case someone else has a similar brian malfunction, but will mark it resolved.

Maliron, 2025-03-03:
Well, I actually take that back. I shouldn't closed it. It seemed like ti was saving it, but now testing further I see it is not. I have to select the tool even after selecting tool 0, hitting save (which sets it as the active spool, which I don't think it should), change spools, then go back to the first spool, tool 0 is no longer selected and must be selected again. I suspect this is because a tool can only be saved to one spool at a time.

[image]

closed 2025-03-03 by Maliron

WildRikku, 2025-03-04:
I can't really follow you what your problem is, but you are correct, that dropdown is irrelevant when your printer only has one spool and one tool. This dialogue is not meant to select the spool for every print.