WildRikku/OctoPrint-SpoolManager #41: QR Code Html Page always shows Spoolname: Beige

Ajimaru, 2025-03-13:
The generated QR Code points to the correct Material but the Spoolname: is always Beige in my case.

[image]

See ID 28 & 16 as en example.

[image]
[image]

Plugin 1.8.0a2
Octoprint 1.10.3
Python 3.10.12

WildRikku, 2025-03-14:
Thanks for the report. I can confirm that the name is not the name of the selected spool. However, it is not "Beige" for me. Do you have a spool that is actually called "Beige"? Might it be the first one you ever created?

closed 2025-03-14 by WildRikku

Ajimaru, 2025-03-14:
> Thanks for the report. I can confirm that the name is not the name of the selected spool. However, it is not "Beige" for me. Do you have a spool that is actually called "Beige"? Might it be the first one you ever created?

Yes one spool color is named Beige, I guess its the first entry in a list ordered by color.

kjsmita6, 2025-05-18:
Happens for me too, but mine is a different label. I think it uses the label of the first spool you entered maybe.

WildRikku, 2025-05-19:
I have fixed the bug with the above commit. The fix will be in the next release.

commit 38ea351f (2025-03-14): Use correct database call in loadSpool(). get_or_none cannot be used directly with an id. Use returned id instead of passed id - this would have made the problem obvious. Remove unnecessary semicolon. Fixes #41