WildRikku/OctoPrint-SpoolManager #45: Security contact

jacopotediosi, 2025-05-14:
Hi @WildRikku,
I'm a security researcher currently working to improve the security of the OctoPrint ecosystem (for example, [here](https://github.com/OctoPrint/OctoPrint/security) you can see some of the vulnerabilities I’ve found in OctoPrint’s core).

Could you please enable [Private vulnerability reporting](https://docs.github.com/en/code-security/security-advisories/working-with-repository-security-advisories/configuring-private-vulnerability-reporting-for-a-repository) for this repository? I'd like to privately discuss a few concerns regarding your OctoPrint-SpoolManager plugin.

I’ve tried reaching out via the email addresses listed on your websites, but they appear to be inactive.
Thanks!

closed 2025-05-14 by WildRikku

Ajimaru, 2025-05-14:
Hi @WildRikku 

Browsing @jacopotediosi repo and commits, he seems legit. What are your concerns?

jacopotediosi, 2025-05-14:
> Hi [@WildRikku](https://github.com/WildRikku)
> 
> Browsing [@jacopotediosi](https://github.com/jacopotediosi) repo and commits, he seems legit. What are your concerns?

It's all ok here, he enabled the private vulnerability reporting