WildRikku/OctoPrint-SpoolManager #49: Variables in display name for tempaltes

turboc1208, 2025-07-30:
I name my spools in a standard format - TYPE-COLOR-SN

Can we have those be selected from the fields selected below.  So I select that the TYPE is petg, Color Silver it changes the display name to PETG-Silver-SN.

Also can we have a running serial number based on the number of spools entered that could be filled in as a sequential number in the display name?

WildRikku, 2025-07-30:
Hmm, variables are an interesting idea, but would be rather complicated to do, since all of those fields might change later. Could you elaborate on what the benefit would be? Why are you using those fields in the spool name? Are you missing the information somewhere? I'm trying to understand if there is something else that could help you.

I'm afraid I don't entirely understand your serial number suggestion. Do you mean have a serial number as a variable? There is already a field for a serial number and there is also #30 which suggests an additional field for a batch number. Also, all spools have a running number for all spools ever entered into the plugin.

turboc1208, 2025-07-30:
How do people traditionally use the display name field?The information is in other fields. I want to use the variables to automatically populate the display name field when i pick the template and fill in those fields. The database id if in a variable would be fine. It’s a way to separate two spools of the same thing.  But that’s just my naming convention. Sent from my iPhone

WildRikku, 2025-07-31:
Since there is no field for the brand name, I use the display name for the specific brand. So I might put Creality as the vendor, fill in material and color and then put "Soleyin" in display name, since Soleyin is the brand and it helps me distinguish it from other Creality PLAs of the same color. That works well with how Spool Manager displays spools, since it adds filament type, color and vendor anyway in all lists.

I used to add the color, too, but since there is the color rectangle, that's actually not that useful as you can see here.

[image]

[image]

turboc1208, 2025-07-31:
> I can see that a running number would be useful. That would require Spool
> Manager to keep track of how often a template was used though. I could,
> however, maybe come up with something that adds a number when a copy is
> created. Similar to file browsers when you copy a file that already exists
> and they name it "File (1)". Would that be useful for you?

I'm fine with it being the database id.  Just a unique number so I can
uniquely identify each spool.  For the short term, can you put the
prospective database id on the add spool form?  Just take the max database
id and add 1.  that should be good enough and you won't have to grab a
value that might not be used if I cancel out of it.